ut-robotics/picr21-team-kobe #1: [Mechanics] - sketches underdefined

Akustav, 2021-09-23:
Sketches for components are underdefined. This creates issues when you need to change model dimensions and the component can react in unexpected ways.
[image]
Can be changed to something like this without issues. It certainly is invalid
[image]

Also:
The warnings in components should not be ignored but fixed:
[image]

This feedback goes for all components you created.

erkiveevali, 2021-09-28:
Resolved the warnings, fixed underdefined sketches where I could find them.

Akustav, 2021-09-28:
Seems fine now.

closed 2021-09-28 by Akustav