ut-robotics/picr21-team-kobe #3: [Mechanics] - spacers in thrower can be touched by ball

Akustav, 2021-09-23:
Spacers inside the thrower might be too close to the ramps outside edge and can collide with the ball
[image]

erkiveevali, 2021-09-28:
The spacers are only between the sidewalls and ramps and not in the wide gap in the center(where the ball would be pushed into the most). Is the ball really gonna be pushed into the thin (7.5mm) gap between on the sides. Also if the ball is compressed on the motor side as well, it won't reach to the spacers.

I could move the spacers further back but I'm afraid that the construction wouldnt be as strong then, if the screw holes are just a couple millimeters from the edge. Another solution would be to move the motor further to front a bit.

Is the gap between the ramps and motor OK right now? If I am gonna need to move components around a bit, then I can change these parameters as well if needed.

Akustav, 2021-09-28:
Okay, true.

closed 2021-09-28 by Akustav