ut-robotics/picr21-team-kobe #5: [Mechanics] - inteference in the assembly

Akustav, 2021-09-23:
There are clipping parts in the assembly that should be looked at. I suspect there is additional interference around the circled areas on the picture. Usually, there should be no interference.
[image]

erkiveevali, 2021-09-28:
Since I changed the components and joints due to other issues then those interference problems should be resolved as well. Fusion  still shows some interference in the assembly, but I'm not entirely sure why.

Akustav, 2021-09-28:
I suggest checking how these 4 corners are cut in the actual model:

[image]

There might be some radiuses that are not accounted for in the design but are there due to milling

erkiveevali, 2021-09-30:
The top ones are fine, since the sides of the ramps arent touching the plate and the corners of the hole in the plate aren't sharp 90degree angles.
On the bottom plate however, the angles do have a small radius that isnt present on the model. I'd say I just file the corners on the bottom plate a bit, to make the ramps fit snugly right against the bottom plate. Thats because I'd like the ramps to be supported by the plate like that.

Akustav, 2021-09-30:
This issue is then OK. You can continue with the CAM

closed 2021-09-30 by Akustav