ut-robotics/picr21-team-kobe #6: [Mechanics] - robot height from the ground

Akustav, 2021-09-23:
How high should the robot be off the ground with its wheels? It is important as it affects how high the motor can be. If it is too high, the ball won't be pulled in and instead bounces off other parts of the chassis. This also allows tuning how low the ramp can be without damaging the carpet or disturbing the robots driving

[image]

erkiveevali, 2021-09-28:
The bottom plate is around 8mm off the ground right now. The thrower ramp extends 5.5mm from the bottom plate - there should be 2.5mm of clearance between the carpet and the thrower ramps.

Akustav, 2021-09-28:
Okay, if it is 8mm then everything seems fine.

closed 2021-09-28 by Akustav