ut-robotics/picr21-team-kobe #29: Refactor states to use enums and matcher

Akustav, 2021-11-06:
An example of how enums can be defined and used in the driving logic instead of a IF-ELSE block:

```
from enum import Enum
class State(Enum):
 FIND = 0
 DRIVE = 1
 AIM = 2
 THROW = 4
```
```
while True:
 ...
 match state:
  case FIND:
   ...
   continue
  case DRIVE:
   ...
   continue
  ...
```

https://github.com/ut-robotics/picr21-team-kobe/blob/9f3d25f5802e43e8fc130219fb4fb0fc0b259346/Robot_runner.py#L33-L38

capnarchie, 2021-11-07:
Is this the correct implementation?
https://github.com/ut-robotics/picr21-team-kobe/blob/f1049b4e73890cd247a7a22e3b0cb84c8893d68e/Robot_runner.py#L14-L18
https://github.com/ut-robotics/picr21-team-kobe/blob/f1049b4e73890cd247a7a22e3b0cb84c8893d68e/Robot_runner.py#L28
https://github.com/ut-robotics/picr21-team-kobe/blob/f1049b4e73890cd247a7a22e3b0cb84c8893d68e/Robot_runner.py#L43-L49

Although I don't see the real difference in this implementation, I'm still using if statements too, or did I misunderstand something?

Akustav, 2021-11-07:
I would use State.FIND instead of State(0) and so on.

This changes main point is to add structure to your code, there is a well-defined set of states with defined behavior in the match state.

You also make code less error prone. For example if you do a typo in the state name then code will run as normal. With enum objects and the state matcher. 

The case `case _:` can be used to handle default cases. You would add an exception there, since it is not normal for the robot to run in an unknown state that might not have been implemented.

closed 2021-11-24 by Akustav