ut-robotics/picr21-team-kobe #31: Target parameter ambigious

Akustav, 2021-11-07:
https://github.com/ut-robotics/picr21-team-kobe/blob/f1049b4e73890cd247a7a22e3b0cb84c8893d68e/Image_processing.py#L15-L23

The target  parameter is very ambiguously defined. Maybe use an enum for basket color. 

Additionally why do you have lHue1, lHue2 and similar? They are gated behind an if-else block so one or the other can be used in one time. I think you can use only one set of these values, unless you are searching for both baskets at the same time.

capnarchie, 2021-11-07:
the target will be a bool for the color from the referee, I'm not intending to threshold 2 baskets at once, the if-else block is for determining which basket thresholding values to use and perform image transformations on it.

Akustav, 2021-11-24:
Since the core of image processing has changed maybe it can be replaced by a Color enum?

capnarchie, 2021-11-29:
removed hsv variables since theyre no longer used and target is defined with color enums now.

closed 2021-12-04 by Akustav