ut-robotics/picr21-team-kobe #74: [Mechanics] - what force holds batteries in the box

Akustav, 2021-11-26:
Question in the title.

erkiveevali, 2021-11-29:
Friction, mainly. Added a fancy latch? to keep the batteries in there as well. A simple stopper made from a screw and a spacer can be added (as a hack) to the front of the battery compartment if the batteries are really that loose in there. But based to my measurements they should fit snugly in there without any fasteners.

Akustav, 2021-11-30:
Would like to see the batteries in the box as well, then it is better to see how they are fastened and that they can not move

erkiveevali, 2021-12-02:
Batteries are now in the box

Akustav, 2021-12-04:
Can't batteries slip out like this? Maybe you can add a bolt head to the top or bottom to act as a stopper
[image]

erkiveevali, 2021-12-06:
Added stoppers to the front as well

closed 2021-12-07 by Akustav