ut-robotics/picr21-team-kobe #80: Use relative numbers based on frame size instead of constants

Akustav, 2021-11-26:
https://github.com/ut-robotics/picr21-team-kobe/blob/86b1d56523d720cfc33aebd9783ef87131fed20b/Robot_runner.py#L88

This is suggested to avoid issues with different resolutions.

capnarchie, 2021-11-29:
done

Akustav, 2021-12-04:
You used:

https://github.com/ut-robotics/picr21-team-kobe/blob/34579003f884aa876f4a892da02aef7702d666d4/Robot_runner.py#L128

This is basically the same thing. I suggest using percentages, like:

`state_data.ball_y > Camera.camera_y * 0,88 (=422)`

You can exclude` Camera.camera_y + 20` since it is impossible for the ball to be outside the frame.

capnarchie, 2021-12-07:
Changed hardcoded resolution dependencies to depend on percentages of resolution instead.

closed 2021-12-11 by Akustav