ut-robotics/picr21-team-kobe #83: Check area instead of 1 pixel

Akustav, 2021-11-26:
https://github.com/ut-robotics/picr21-team-kobe/blob/86b1d56523d720cfc33aebd9783ef87131fed20b/image_processor.py#L136

Instead of sampling one point, sample a area to get a less noisy result and increase accuracy.

capnarchie, 2021-11-29:
Done although accuracy is pretty much same for both except for longer distance.

Akustav, 2021-12-04:
Accuracy contains of both direction and speed. Is the aim direction good/bad? Overall I would check the stability of the measurement when the robot aims at a static object.

capnarchie, 2021-12-07:
Well it seems good enough that I can not differentiate between the two, so I guess it's working as intended.

Akustav, 2021-12-11:
One thing you can change for robustness is in case the range gets an out of bounds exception you can fallback to the one-pixel solution. If you do not want to double-check the ranges:

https://github.com/ut-robotics/picr21-team-kobe/blob/1c09a2ff8e201bfb6875c36d273c914e2959c723/image_processor.py#L146-L148

closed 2022-01-25 by Akustav