ut-robotics/picr21-team-kobe #88: [Mechanics] - Wheel holder bottom plate tab spacing off

Akustav, 2021-11-30:
The spacing should be 60 but for you it is 58,966. This creates some interference issues:
[image]

erkiveevali, 2021-12-03:
[image]
The dimensions on the sketch are correct, and I am using circular pattern to create 3 identical mounts. Im not sure where this weird number 59.966 comes from

Akustav, 2021-12-04:
The issue comes from the fact that  this line is not horisontally/vertically aligned. So it is at an angle and that creates a shift
[image]

erkiveevali, 2021-12-06:
All the design issues should be resolved now.

closed 2021-12-07 by Akustav