ut-robotics/picr21-team-kobe #90: [Mechanics] - potential recreation of issue with thrower

Akustav, 2021-12-07:
If I remember correctly there is an issue with your current thrower that does not center the ball to well. When the ball rolls into the robot depending on which side it hits first, the ball gets grabbed and thrown differently. The areas in question:
[image]

I would discuss it with your programmers. One solution would be to move the thrower sidewalls closer in the front or move the motor back a little so the ball has some time to calm down before it is pulled into the thrower motor.

erkiveevali, 2021-12-08:
Discussed the issue with Uku and we think that the issue on the old robot is mostly caused by the fact that the thrower sidewalls (especially one of them) dont fit the bottom plate that well. The cause of that problem is the fact we needed to modify the parts after milling to get them to fit at all. The whole test thrower moved to the front and created those corners that interfere with the ball. Although I cut the interfering corners, the current solution isnt ideal at the moment. As long as the new parts fit together like do in the model, that shouldnt be as big of an issue as it is now.

Moving the thrower inwards could yield better results in giving the ball time to calm down but it also means the robot has to move "blindly" forward more for the ball to reach the motor. So that could cause other problems. 

Also moving the thrower inwards as much as 2cm will take but most of the free space behind the thrower (which was initially meant for motor wiring etc to fit into) leaving very little room inside the robot.

So I am not against moving the thrower inwards but I am not sure how much better results that would give overall

Akustav, 2021-12-08:
Yeah, I agree that it is a tradeoff and your decision in the end. Some things to consider:

As for moving the thrower inside, I also took into account that the ramp can be slimmed down and moved without much effort. Size of cabling on the motors is quite minimal, in my opinion. As for the blindly part the robot aims towards the basket at all times so it should not affect accuracy when the visual on the ball is lost. Overall I would try to achieve as robust a physical design as possible to allow for maximum accuracy. Every missed shot costs a lot of time and gives opponents an additional oppurtunity.

erkiveevali, 2021-12-08:
[image]
Was able to move the motor inside 11mm without redoing the ramps or thrower sides. Hopefully its better that way.

Akustav, 2021-12-08:
I would add something like this as well:

[image]

It strengthens the ramp again and also supports it by resting the piece on the top and bottom plate

erkiveevali, 2021-12-08:
Reinforced the ramps as described on the picture

closed 2021-12-08 by Akustav