ut-robotics/picr21-team-kobe #91: [Mechanics] - small circles in CAM only connected with one tab

Akustav, 2021-12-07:
As can be seen small holes are held in with only 1 tab:

[image]

erkiveevali, 2021-12-08:
Decreased the minimal distance between tabs, now the small holes have 2 tabs.

Akustav, 2021-12-08:
Another potential solution you might want to think about is making a separate procedure for these small components. This way you save time on cutting and finishing of the tab residue on the components.

erkiveevali, 2021-12-08:
The smaller holes already have a separate 2D contour under the setup. You mean getting rid of the tabs on these small cutouts?

Akustav, 2021-12-08:
Some unintended tabs have appeared? You can also manually place tabs.
[image]

erkiveevali, 2021-12-08:
Placed the tabs manually now, unintended tabs should be gone

closed 2021-12-08 by Akustav